kbeckmann/PicoCart64 #2: Investigate RP2040 XIP/QSPI from second RP2040

kbeckmann, 2022-06-24:
Investigate if it's possible and practical to let one RP2040 (pico1) boot from a second RP2040 (pico2) by connecting the Flash pins to the gpios of the second.

Might need to use PIO to emulate flash in the second RP2040.

Will probably need to lower the flash speed configuration

Implementation idea:

pico2 loads the image it wants to emulate into RAM first.
pico2 sets up PIO to quickly serve QSPI read requests.
Could borrow code/ideas from the PicoCart64 code.

swetland, 2022-07-05:
Luke Wren mentioned (in a twitter conversation) that the bootrom's download of boot2 (256 byte secondary loader) is done at 1MHz and believes even a bit-banging solution should be workable.

Rather than trying to emulate QSPI modes or faster transports, my thinking was to just write a custom boot2 that would use SPI to pull an image into SRAM and jump to it.

For my use cases (using the 2nd RP2040 as an IO expander, etc), running a fixed image that is loaded at boot and fits inside the (generous) 264KB of onboard SRAM would be plenty sufficient.

At 8MHz and using large transfers, SPI download of a 128KB image would probably take under 150ms. (131ms if there were no overhead between transfers and all data were sent back to back)

kbeckmann, 2022-07-05:
Another approach which was discussed in our discord was to physically share one flash chip between the two RP2040 - even less work!

The reset/run pin of MCU1 is pulled low with a resistor, and is also connected to MCU0.

MCU0 boots and loads its program into SRAM. It reads the GPIO connected to MCU1's reset pin and sees a '0' - this lets it know that it is MCU0.

MCU0 now enables the GPIO, letting MCU1 boot. When MCU1 has booted and loaded its program into SRAM, it reads the same GPIO pin, but we make sure to connect it to a net that is pulled high in idle - now it knows it is MCU1.

kbeckmann, 2022-07-16:
We have to verify the following:

We probably need to configure the CS line to be open drain and enable a pull-up. Otherwise, when the other rp2040 is idling, it will keep CS=1 which will interfere when the first pico wants to read from the flash.
If that's not possible (because of bootrom etc), the boot sequence probably needs to be like this:

Pico1 boots normally, copies its program to RAM
Pico1 disables its SSI interface, ensuring that CS line becomes high impedance / disable output
Pico1 pulls Pico2_RUN low, allowing Pico2 to boot
Pico2 can now assert FLASH_CS high or low without any issue.
Pico2 disables its SSI interface

FLASH_CS is now allowed to be asserted by either Pico, high or low.

kbeckmann, 2022-07-24:
[image]

Proof of concept works!

```c

/**
 * SPDX-License-Identifier: BSD-2-Clause
 *
 * Copyright (c) 2022 Konrad Beckmann
 */

#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/structs/ssi.h"
#include "hardware/structs/ioqspi.h"

int main(void)
{
	int count = 0;

	// GPIO 2 is connected from Pico 0 to the RUN pin of Pico 1
	gpio_init(2);
	gpio_set_dir(2, GPIO_IN);

	// GPIO 25 / LED is a slow signal to show if we are running
	gpio_init(25);
	gpio_set_dir(25, GPIO_OUT);

	int pico_id = gpio_get(2) + 1;
	stdio_init_all();

	sleep_ms(1000);

	// Turn off SSI
	ssi_hw->ssienr = 0;

	// Set all QSPI IOs high impedance (disable OE)
	for (int i = 0; i < 6; i++) {
		ioqspi_hw->io[i].ctrl = (ioqspi_hw->io[i].ctrl & (~IO_QSPI_GPIO_QSPI_SCLK_CTRL_OEOVER_BITS) |
								(IO_QSPI_GPIO_QSPI_SCLK_CTRL_OEOVER_VALUE_DISABLE << IO_QSPI_GPIO_QSPI_SCLK_CTRL_OEOVER_LSB)	);
	}

	sleep_ms(1000);

	if (pico_id == 1) {
		// I am Pico 1, let Pico 2 boot
		gpio_set_dir(2, GPIO_OUT);
		gpio_put(2, 1); // GPIO2 = HIGH
	}

	while (true) {
		count++;
		printf("Hello, world! I am Pico %d -- %d\n", pico_id, count);
		gpio_put(25, !!(count & 1)); // GPIO25 = toggle LED slowly
		sleep_ms(1000);
	}
	return 0;
}

```

closed 2022-07-24 by kbeckmann