kbeckmann/PicoCart64 #15: Integrate FreeRTOS

kbeckmann, 2022-07-05:
We need an RTOS to run CIC emulation and EEPROM + FlashRAM emulation on the same core.

Investigate if there is a FreeRTOS port for RP2040 out there already, otherwise port it.

gcochard, 2022-07-05:
There's a ChibiOS port for the rp2040 which could be viable: https://forums.raspberrypi.com/viewtopic.php?t=308510

kbeckmann, 2022-07-05:
Thanks! I haven't looked closer at ChibiOS. How does it compare to FreeRTOS feature wise? I think we mostly only need some sort of scheduler that runs on one of the cores that lets us run multiple tasks in parallel with preemption. We probably don't "proper" multicore support, since the core that handles the N64 PI interface has absurdly high latency requirements, so it will be fully busy with that.

kbeckmann, 2022-07-10:
Just noticed that the pico_w examples include a FreeRTOS integration too: https://github.com/raspberrypi/pico-examples/blob/master/pico_w/freertos/CMakeLists.txt

kbeckmann, 2022-07-10:
Integrated FreeRTOS now.

commit 4197cefc (2022-07-10): sw: Integrate FreeRTOS and add basic tasks

Closes #15
closed 2022-07-10 by kbeckmann