kbeckmann/PicoCart64 #23: [WiFi] Add server solution to push out ROMs to carts

kbeckmann, 2022-07-12:
Let the cart connect to the user's wifi, then connect to a server (either local or on the internet maybe?), then let the user control the server to push out a ROM file on demand.

clktmr, 2023-10-02:
Will it be possible to access the Wifi module from the ROM?

kbeckmann, 2023-10-02:
As of now, v2 is scrapped. I'm working on adding wifi support to v1 lite with a pico-w. It should be possible yes but there is no API yet.

clktmr, 2023-10-03:
Will the Picos SPI be memory mapped on the PI bus? Or how will this work?

kbeckmann, 2023-10-03:
Yes this is how it works today - the N64 binary / ROM is stored on the pico's external spi flash. When there are PI read requests in the ROM address range, it will load data from the external flash (with or without compression). See https://github.com/kbeckmann/PicoCart64/blob/develop/sw/picocart64_v1/n64_pi_task.c#L91-L118

closed 2023-12-08 by kbeckmann