kbeckmann/PicoCart64 #47: can't run testrom

hpx1984, 2023-05-07:
Hi Sir

built one lite ,can boot to the menu but can't run testROM(there has one fail item)
[image]
pressed all button and no action 
[image]

My step as below, please help me for improve it, many thanks Sir
1. programmed picocart64-v1-ntsc.uf2
2. Created testrom.uf2
3. programmed testrom.uf2

kbeckmann, 2023-05-07:
You have successfully booted the test rom actually. The failed test is a persistent SRAM test - if you press the reset button it will pass. 

The menu is just a mock menu. It doesn’t do anything.

closed 2023-05-07 by kbeckmann

hpx1984, 2023-05-07:
many thanks your quickly replied ,Sir

is it meets this lite version still can't run any ROMs or just can run the ROMs which show in the mock menu ?

i want to buy a Weact Board if it can support to run the ROM

kbeckmann, 2023-05-07:
Follow the steps in https://github.com/kbeckmann/PicoCart64/blob/develop/sw/README.md or use https://picoc.art to change the rom

hpx1984, 2023-05-08:
thank you so much ,sir
My 16m board will arrive end of this week, so happy time on your project, so great project