kbeckmann/PicoCart64 #51: Compatipel qith that pico?

Liriel-666, 2023-12-24:
[image]
Is the lite version of the board compatible with that pico 2040?

kbeckmann, 2023-12-25:
No, not without modifications. Notice the pins vin and vout, and compare with the original or weact pico board.

Liriel-666, 2023-12-25:
is Vin and Vout the only difference?
sry i am not that big in the pico thing and for me the rest looks good

Liriel-666, 2023-12-25:
to what i can see the pinout ist most the same to the non weact pico. 
i see some difference  in the last 10 pins but i dont now how to fix it.
What pin of the lite  1.4 board must connect to what pin on the black pico board?

[image]
[image]

curcius, 2024-01-17:
Were you able to use this card? I bought the same one and did all the soldering points but it doesn't generate an image on the N64 =[

ppattss, 2024-01-18:
> Updating, I was able to run Homebre as Testrom and a cell phone game converted to UF2, but the Nintendo 64 games don't start at all. I've checked pin by pin and there are no shorts or soldering failures.

It might be related to region-checking and not having the correct one selected when making the UF2 file? or CIC issue?

Did you reconnect VREF to GP29? the VREF jumper isnt connected on the board i got, but is in all the pics of that board.

curcius, 2024-01-18:
> It might be related to region-checking and not having the correct one selected when making the UF2 file? or CIC issue?
>
> Did you reconnect VREF to GP29? the VREF jumper isnt connected on the board i [...]

Thank you for responding, but I've tried everything and neither UF2 NTSC nor PAL M works, I've tested several games and nothing, I've closed and opened the jumper and it gives the same results... I believe it's some incompatibility with the flash, no. I can think of nothing else, the rpi works, I've tested it with other programs like Micropython, but 64 games don't run, just homebrew...

curcius, 2024-01-19:
I managed to solve it, I had to use the 3v3 from the console board directly on the RPI, I was using the RPI's USB power supply and it didn't work, I made the jumper to the 3v3 on the RPI and it worked perfect. Thank you everyone, I hope the project moves forward. 
[image]

Liriel-666, 2024-01-19:
i dont get my to work with a weact pico
[image]

I solder all correct. because i destroy on unsolderng the other pico i replace the jumper with a cable
al solderpoints are there and correct. 
i even solder the debug port

nothing with any uf2 convertet file.
I put firmware on it and then restart it. 
then i put the testrom or a game on it and nothing on my pal n64

samowitsch, 2024-07-15:
@Liriel-666 i have the exact same problem as you. I ordered like you a WeAct RP 2040 and nothing is working. Did you solved it in the meantime?

samowitsch, 2024-08-07:
@kbeckmann Neither the WeAct 16MB nor the Bitfunx 16MB Pico board works. Are there any secret tips that aren't mentioned anywhere? Does the picocart64-v1-pal/ntsc.uf2 work with any Flash Ram size?

kbeckmann, 2024-08-08:
> @kbeckmann Neither the WeAct 16MB nor the Bitfunx 16MB Pico board works. Are there any secret tips that aren't mentioned anywhere? Does the picocart64-v1-pal/ntsc.uf2 work with any Flash Ram size?

I'm sorry, I don't know to be honest. WeAct 16MB have been known to work earlier, but it's also possible that the manufacturer changed something, e.g. maybe the flash is not the 133MHz variant or so?

samowitsch, 2024-08-12:
The following "solutions" work for me. (By the way, thanks for the feedback on Discord from be0w0lf710, which pointed me in the right direction. I would never have questioned BSS84 that something is not working properly here.)

**Bitfunx board (100% pin compatible)**
[image]

**Weact board**
[image]

reflekt00, 2024-10-22:
@samowitsch has this setup bridging 5V on the weact board to the 3.3V rail been working reliably for you? I'm in the same boat, the cart won't load a rom with the transistor in place.

samowitsch, 2024-10-26:
@reflekt00 it worked for me without the components and just a jumper wire. But with this „workaround“ the USB cable should not be connected when the picocart is plugged into the N64.

closed 2024-12-15 by kbeckmann