kbeckmann/PicoCart64 #52: Can't find picon64 firmware

kevincondrick, 2024-03-29:
So I just got the picon64 v1 lite cart off eBay and now everything I see on here says it needs to be flashed with the firmware. The link I found to it is https://nightly.link/kbeckmann/PicoCart64/workflows/20_build/develop/picocart64 
 
I keep getting and error 404 when I click on it idk if anyone has this same issue or what but need help

Be0w0lf710, 2024-03-29:
The build is over 90 days old which is why the link is no longer active.  You will need to build it yourself from what i can tell.

kbeckmann, 2024-03-29:
Link should work again.

closed 2024-03-29 by kbeckmann

kbeckmann, 2024-03-29:
Created a release from the last develop commit with an attached zip, so it should stay up forever. The 90 day timeout is annoying.

https://github.com/kbeckmann/PicoCart64/tags

kevincondrick, 2024-03-29:
Thanks man!!!