kbeckmann/PicoCart64 #54: USB Debug functionality

markusbkk, 2024-05-21:
Curious.

Are there any plans to add the ability to stream data straight to the cartridge over the USB port while everything is plugged in and the console is running?

Would be a boon for homebrew devs looking for a cheap USB devcart.

Carl-Llewellyn, 2025-03-29:
This is possible by setting up the USB device in main() using stdio_init_all() after overclocking, then printf can be used to print to USB. 

Then if you write to the PC64_REGISTER_UART_TX from the ROM/N64 and replace the stdio_uart_out_chars() with printf() it will come in over the USB instead of UART.

Not sure about writing to N64 memory at the mo.

Carl-Llewellyn, 2025-03-31:
To write to N64 memory:

Set up some buffers:

```
static uint8_t __attribute__((aligned(16))) usb_buffer[USB_BUFFER_SIZE] = {0};
uint32_t usb_bytes_received = 0;
uint32_t read_word = 0;
```

Set up a usb task with FreeRTOS:

```
void usb_task_entry(void *pvParameters) {

  while (true) {
    while (!tud_cdc_connected()) {
      vTaskDelay(pdMS_TO_TICKS(100));
    }

    if (tud_cdc_available()) {
        usb_bytes_received = tud_cdc_read(usb_buffer, USB_BUFFER_SIZE);
		if (usb_bytes_received >= sizeof(uint32_t)) {
 		   read_word = *(uint32_t *)usb_buffer;
		}
    }
	
    vTaskDelay(pdMS_TO_TICKS(10));
  }
}
```

Add a USB RX register:

```
#define PC64_REGISTER_UART_RX 0x00000010
```

Add the register to this switch and then read it into next_word:

```
          switch (last_addr - PC64_CIBASE_ADDRESS_START) {
          case PC64_REGISTER_MAGIC:
            next_word = PC64_MAGIC;
            break;
          case PC64_REGISTER_FLASH_JEDEC_ID:
            next_word = g_flash_jedec_id;
            break;
          case PC64_REGISTER_UART_RX:
            next_word = read_word;
            break;
          default:
            printf("DEFAULT");
            next_word = 0;
          }
```

Should be good to go.

I'll probably make a fork for this but wont maintain it as I'll just be using it for a project of my own. Lmk and I'll link it here if I do it!