kbeckmann/PicoCart64 #56: Problem with ROMs whose size isn't a multiple of 1024 bytes

michalderkacz, 2024-07-22:
The compression algorithm in the JS function `generateAndSaveUF2` doesn't support the ROM size that isn't a multiple of 1024 bytes.

This is the likely cause of the problem with [N64 240p test suite ROM](https://www.dropbox.com/s/76skpu3v4q7twjb/240pSuite-N64-Alpha-1.zip?dl=0) that I reported before whose size is 6421760 bytes = 6271.25 KiB.

The `generateAndSaveUF2` calculates the number of chunks as `num_chunks = Math.floor(rom.fileSize / chunk_size)` where `chunk_size = 1024` so it doesn't handle the last incomplete chunk in any way.

michalderkacz, 2024-07-22:
Another bug I found is using `Uint16Array` that produce native-endian byte stream. It's unlikely that someone will open the converter code on a big-endian machine but it's still a bug.

I'm translating your JS code to Go so such bugs easily reveal themselves.

kbeckmann, 2024-07-22:
Thanks for finding the root cause. If you have the time to create a PR, I’d be happy to merge a fix in the repo https://github.com/kbeckmann/PicoCart64/tree/gh-pages