kbeckmann/PicoCart64 #62: Not Work (N64 - Revision 03)

xmarciox, 2024-09-18:
I used an F-1 World Grand Prix cartridge as a donation and only kept the 4kb save chip.
However, there is no signal in the game, I have already tested all the contacts on the cartridge, there are no broken tracks, I have recorded the firmware several times and nothing, not even the test ROM works.

What could I be doing wrong?

My Nintendo 64 board model is 03.
I've already tried powering directly via the USB-C peak and nothing either.

[image]

kbeckmann, 2024-09-18:
> not even the test ROM works

Start there and get the test rom to work first. Check your soldering and signal integrity.

xmarciox, 2024-09-18:
> > not even the test ROM works
> 
> Start there and get the test rom to work first. Check your soldering and signal integrity.

I tried running the test rom several times and was unsuccessful.
I measured each point from the peak to the plate pad and they are all ok. I spent several hours trying and nothing.
Not what could be wrong.

kbeckmann, 2024-12-15:
I updated the build environment and made stable releases end of October, please try with those. If those don’t work, the root cause is probably a hardware problem.

Carl-Llewellyn, 2025-03-29:
I've done the same and it works fine for me (didn't realise anyone else was doing this janky set up haha!). 

I've found that if you're grounding is loose it doesn't like to work, or it works very intermittently. So, worth a go if you are sure you're hooked up correctly.