kbeckmann/PicoCart64 #64: Pico cart savings settings

kevincondrick, 2024-12-15:
I'm going to start by saying I know that it needs that one chip to save setting. I have a one that has the save chip and one that does not. So I put a mod of 007 GoldenEye in the one without the save chip and it works but does not save settings, so the volume and sound effects always need reset every time. Then I put the same rom on the one that has the save chip and it doesn't work at all. Any ideas why that is maybe a firmware issue?

kbeckmann, 2024-12-15:
There are many details missing here. Please describe exactly which hardware you’re using. Also consider asking in the discord since there are some more people there who could help you out.

kevincondrick, 2024-12-15:
Idk I'm using a pico N64 cart I bought off eBay and it has the Bu9850 chip on it and another one that doesn't.

closed 2025-04-04 by kbeckmann